rkaczorek/astroberry-diy #3: rpi_focus cmake error

jakugab, 2018-07-02:
Dear Radek!
First of all, I'd like to say a great thanks for your work on Astroberry Server and Astroberry DIY.
I have an RPI 3 model B+ with Raspbian, and I wanted to install the Astroberry focuser with cmake, following the installation instructions, but I always got this error message:

Scanning dependencies of target indi_rpifocus
[ 93%] Building CXX object CMakeFiles/indi_rpifocus.dir/rpi_focus.o
/home/pi/astroberry-diy/rpi_focus.cpp: In member function :
/home/pi/astroberry-diy/rpi_focus.cpp:212:73: error:  was not declared in this scope
        SetFocuserCapability(FOCUSER_CAN_ABS_MOVE | FOCUSER_CAN_REL_MOVE);
                                                                        ^
CMakeFiles/indi_rpifocus.dir/build.make:62: recipe for target 'CMakeFiles/indi_rpifocus.dir/rpi_focus.o' failed
make[2]: *** [CMakeFiles/indi_rpifocus.dir/rpi_focus.o] Error 1
CMakeFiles/Makefile2:215: recipe for target 'CMakeFiles/indi_rpifocus.dir/all' failed
make[1]: *** [CMakeFiles/indi_rpifocus.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

Earlier the Astroberry Focuser worked perfectly in Kstars installed on my RPI2 with Ubuntu Mate, but unfortunately the SD card injured, and the whole setup destroyed. I don't remember, how I installed the working focuser driver, but when I tried to compile the driver, I got the same error message.
Please help me to solve this problem! (I am a really-really end-user :-) ).

Thanks in advance
Gabor

rkaczorek, 2018-07-03:
Hi Gabor. Apparently core INDI libraries have been changed since last code review. I need to synchronize the sources to the latest INDI. As a workaround you can safely remove the line:

SetFocuserCapability(FOCUSER_CAN_ABS_MOVE | FOCUSER_CAN_REL_MOVE);

from rpi_focus.cpp (line 212).
The driver will compile without errors.

I will keep this issue open until I update the code.

jakugab, 2018-07-04:
Hi Radek!
I removed the line from rpi_focus.cpp, and I compiled the driver without error. It works perfectly with RPI2 + ubuntu mate + Kstars, RPI2 + Asrtoberry Server and with RPI3+ + Raspbian + libindi1 setup controlled by Kstars Lite. Thank you very much for your help! 😁
Gabor

rkaczorek, 2018-12-21:
Fixed in the current version

closed 2018-12-21 by rkaczorek

jakugab, 2018-12-21:
Hi Radek! 
Thank you again!
Gabor