rkaczorek/astroberry-diy #8: error: ‘FOCUSER_CAN_REVERSE’

greg-the-maker, 2019-05-04:
Hi,

I have this error when compiling.

INDI 1.7.4

> [ 66%] Built target libbcm2835
[ 75%] Building CXX object CMakeFiles/indi_rpifocus.dir/rpi_focus.o
/home/pi/astroberry-diy/rpi_focus.cpp: In constructor ‘FocusRpi::FocusRpi()’:
/home/pi/astroberry-diy/rpi_focus.cpp:139:66: error: ‘FOCUSER_CAN_REVERSE’ was not declared in this scope
  FI::SetCapability(FOCUSER_CAN_ABS_MOVE | FOCUSER_CAN_REL_MOVE | FOCUSER_CAN_REVERSE); // | FOCUSER_CAN_ABORT);
                                                                  ^~~~~~~~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp: In member function ‘virtual bool FocusRpi::initProperties()’:
/home/pi/astroberry-diy/rpi_focus.cpp:264:2: error: ‘FocusMaxPosN’ was not declared in this scope
  FocusMaxPosN[0].min = 1000;
  ^~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp: In member function ‘virtual bool FocusRpi::ISNewNumber(const char*, const char*, double*, char**, int)’:
/home/pi/astroberry-diy/rpi_focus.cpp:377:4: error: ‘FocusMaxPosN’ was not declared in this scope
    FocusMaxPosN[0].min = (int) FocusMaxPosN[0].min * resolution / last_resolution;
    ^~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp:381:17: error: ‘FocusMaxPosNP’ was not declared in this scope
    IDSetNumber(&FocusMaxPosNP, nullptr);
                 ^~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp:403:19: error: ‘FocusReverseSP’ was not declared in this scope
    deleteProperty(FocusReverseSP.name);
                   ^~~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp: In member function ‘virtual bool FocusRpi::saveConfigItems(FILE*)’:
/home/pi/astroberry-diy/rpi_focus.cpp:523:29: error: ‘FocusMaxPosNP’ was not declared in this scope
     IUSaveConfigNumber(fp, &FocusMaxPosNP);
                             ^~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp:525:29: error: ‘FocusReverseSP’ was not declared in this scope
     IUSaveConfigSwitch(fp, &FocusReverseSP);
                             ^~~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp: In member function ‘virtual IPState FocusRpi::MoveAbsFocuser(int)’:
/home/pi/astroberry-diy/rpi_focus.cpp:606:6: error: ‘FocusReverseS’ was not declared in this scope
  if (FocusReverseS[REVERSED_ENABLED].s == ISS_ON) {
      ^~~~~~~~~~~~~
/home/pi/astroberry-diy/rpi_focus.cpp:606:20: error: ‘REVERSED_ENABLED’ was not declared in this scope
  if (FocusReverseS[REVERSED_ENABLED].s == ISS_ON) {
                    ^~~~~~~~~~~~~~~~
CMakeFiles/indi_rpifocus.dir/build.make:62: recipe for target 'CMakeFiles/indi_rpifocus.dir/rpi_focus.o' failed
make[2]: *** [CMakeFiles/indi_rpifocus.dir/rpi_focus.o] Error 1
CMakeFiles/Makefile2:104: recipe for target 'CMakeFiles/indi_rpifocus.dir/all' failed
make[1]: *** [CMakeFiles/indi_rpifocus.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

>

greg-the-maker, 2019-06-09:
I still can't get it to compile. Some assistance would be most welcome.

rkaczorek, 2019-06-10:
What is your version of libindi? I have just recompiled with the latest 1.7.8 and no problem whatsoever.

greg-the-maker, 2019-06-11:
Hi,

It's 1.7.4 downloaded from https://indilib.org/download/raspberry-pi/download/6-raspberry-pi/9-indi-library-for-raspberry-pi.html

rkaczorek, 2019-06-11:
You need a latest version to compile. If you cannot install it from binaries you need to compile the latest INDI from https://github.com/indilib/indi

greg-the-maker, 2019-06-11:
That worked! Thank you :)

Odd that the instructions for the raspberry pi install say to run the 1.7.4 version?

closed 2019-06-11 by greg-the-maker