rkaczorek/astroberry-diy #9: support nanopi-m4

dragonlost, 2019-08-12:
hello.

I would like to use to run your indi driver on my nanopi-m4.
There is a library wriringpi that works pretty well.
Unfortunately I have no gpio accessible via wiring pi on the pins 29, 31, 33 and 35 (the pins of the part relay)

On the other hand I have pins 8, 10, 12, 22 and 26 which are accessible (with numbers 15, 16, 1, 6 and 11 on wiring pi).

dragonlost, 2019-08-19:
Hello.

How should I modify the program to change the pins used for the relay part?

rkaczorek, 2019-08-19:
Pin mapping for wiringpi is defined in lines 52-55 of rpi_brd.cpp
Possibly you can use the pin numbers as you already identified in the first post.

rkaczorek, 2020-03-02:
PIN mapping is now configurable from UI

closed 2020-03-02 by rkaczorek