rkaczorek/astroberry-diy #11: AbortFocuser() handling

rkaczorek, 2020-04-02:
I need help in implementing AbortFocuser() function in the driver.
I'm missing basic concept how it should be handled according to INDI API.

jryd2000, 2020-09-10:
You may want to look at my fork ([https://github.com/jryd2000/astroberry-diy]) for the AbortFocuser functionality. Unfortunately, I changed a bunch of the code to support driving a 28BYJ-48 via a ULN2003 motor driver (without keeping the 2 drivers you already implemented, I realized later I probably could have extended it).  After I did that though, I wanted to get the AbortFocusert working.  I had to remove all of the msleeps that were blocking other threads and instead utilized the SetTimer to facilitate moving to each success tick.  This allows for the abort function to be executed while waiting for a successive tick and cause it to stop it.  

Also thanks for these drivers, as well as the Astroberry distribution, I am using it at home with my NexStar 6SE.

rkaczorek, 2020-09-10:
Thanks mate! I will give it a try

picciux, 2021-09-08:
Hi Radek,

please see my PR #20 for a completely different approach.
Bye

commit a4f3c250 (2021-11-11): Abort focuser motion added. Fixes #11
closed 2021-11-11 by rkaczorek