rkaczorek/astroberry-diy #13: Waveshare Stepper Motor Hat focuser support

luzik, 2020-12-15:
It would be nice to have it supported, as it looks like natural Astroberry companion :). This board is quite cheap (compared to other hats) and should support big range of motors.

rkaczorek, 2020-12-15:
Kevin Rose has already developed a driver for this hat, based on my [Adafruit Motor HAT](https://github.com/rkaczorek/astroberry-amh). Take a look at [this thread](https://indilib.org/forum/focusers-filter-wheels/6038-indi-focuser-driver-for-waveshare-stepper-motor-hat-for-raspberry-pi.html) and Kevin's driver [source](https://github.com/k-ross/indi-wmh-focuser).
There are some limitations of the waveshare device so I'm not sure if I want to support it with Astroberry DIY driver. Some details have been discussed [here](https://indilib.org/forum/focusers-filter-wheels/6038-indi-focuser-driver-for-waveshare-stepper-motor-hat-for-raspberry-pi.html?start=84#60858).

luzik, 2020-12-15:
Thank you for clarification. If you ware consider microstepping control as limitation, IMHO its not an issue, because it should be only used for smooth move, and not for resolution multiplier. For this case it can be set once per project.
Pleas look [here](https://hackaday.com/2016/08/29/how-accurate-is-microstepping-really/)
What others limitation can You see ? I've just ordered one, so maybe I will try help here.