rkaczorek/astroberry-diy #15: Astrobery Relays On/Off seem to be the opposite

AnakChan, 2021-02-09:
I've just configured the Astroberry Relays for the ASIAir Pro DC Hub and the On/Off toggle switches seem to be opposite to what it is. i.e. On switches off the port, Off switches on the port.

[image]
[image]

For now I switched the following and recompiled and it seems to reflect correctly that Off is now switching off the DC port, and On is switching on the DC port :-

gpiod_line_set_value(gpio_relayX, 1);
gpiod_line_set_value(gpio_relayX, 0);

AnakChan, 2021-02-10:
Update: Please see KenS's fork adding a parity :-

https://github.com/ken-self/astroberry-diy/tree/relays

[image]

rkaczorek, 2021-02-22:
I can get it to the upstream if I get pull request. As for now KenS's repo is not available (404)

rkaczorek, 2021-03-06:
Active state LOW/HIGH added in version 2.7

closed 2021-03-06 by rkaczorek