rkaczorek/astroberry-diy #16: Support for ULN2003

flipkill1985, 2021-07-13:
Hello,

does this work with driver ULN2003?
If not can you implement this?

best regards Jan

rkaczorek, 2021-07-14:
No it doesn't work with ULN. To be honest I don't think it is a good idea to use this low level IC for stepper motor control with rasperry pi. It uses 7 GPIOs and needs complete software based implementation of stepping. In my opinion it's an overkill. Why would you like to use it?

astrojolo, 2021-07-18:
Hi Radek,

I think it may be a reasonable feature since ULN2003 is enough to control standard unipolar geared steppers like the ones in Moonlite or Robofocus solutions. I may try to implement this - just let me know if I should work with the branch and pull request or do you prefer to fork the repo.

Best regards,
Lucas

rkaczorek, 2021-07-18:
Hi Lukasz,

Are we talking about [this IC](https://www.st.com/en/interfaces-and-transceivers/uln2003.html)? This is a bare metal IC containing 7 Darlington arrays. I assume you would like to use just 4 of them and control stepper coils with it. Pretty straight forward however you cannot use stepper library on raspberry pi the same way as on arduino. As the result you need to code whole procedure from scratch, including step sequence etc. I'm ok with pull request but to be honest I'm not sure you can easily integrate code needed for ULN2003 with this Astroberry DIY driver, or I'm missing something obvious. If you want to go this way let me know if you need any help

jryd2000, 2021-07-21:
You can check out my fork at https://github.com/jryd2000/astroberry-diy to see a ULN2003 implementation.  Unfortunately to make it easier to code I removed the other 2.  I also made some changes to implement aborts.

rkaczorek, 2021-07-21:
Well done! This shows however that integrating it to the existing driver is not a piece of cake and takes a lot of work. I will think it over how it can be done so maybe we'll come back to this topic some day