rkaczorek/astroberry-diy #17: Request: Support Wireless connection with Stepper moter

jctk, 2021-08-11:
Hello.
Astroberry Focuser is great tool for me. It has been an indispensable tool for my astronomical observations.

Now, I have a request for Astroberry Focuser.
Could you possible to make the connection to the motor wireless ?
Perhaps this can be achieved by supporting remote GPIO. Remote GPIO can be supported by pigpio.

For example, you can install a Stepper motor and handler HAT in a Raspberry PI Zero W, and install an Astroberry Focuser Driver rewritten in pigpio in a Raspberry PI 4.
The Raspberry PI Zero W should be connected to the Raspberry PI 4 via Wi-Fi or Bluetooth PAN. 
Controlling the focusing module with Ekos on Raspberry PI 4 would allow the Stepper motor on Raspberry PI Zero W to work.

Thanks

rkaczorek, 2021-08-13:
Hi, I believe you should use different strategy for achieving your goals.
My suggestion is to run INDI server with Astroberry Focuser driver on Raspberry Pi Zero and autostart & autoconnect the driver. Raspberry Pi Zero should be connected to a network. Then configure a remote driver on other device which runs KStars & Ekos using Astroberry Focuser@IP_ADDRESS, where IP_ADDRESS identifies your Raspberry Pi Zero.
This way your focuser becomes wireless a no change is required to the source code ;)

jctk, 2021-08-13:
I think your explanation is more INDI-like and smarter than my idea.
Thank you, I understand very well.

closed 2021-08-13 by jctk