rkaczorek/astroberry-diy #19: Saving and Loading Multiple Named Configuration Files

jctk, 2021-08-25:
Could you please implement the ability to save and load different settings into different named settings files?

I am using two different focusers on one Astroberry DIY focuser. Only one of the two focusers can be connected at a time.

The Astroberry Focuser requires different settings for each focuser. This is because the specifications for Reverse motion, Max. Position, Max. Travel, etc. of the two focusers are different.

We believe that if the Astroberry Focuser can handle multiple configuration files, it will greatly simplify and improve the convenience of the settings for me.

rkaczorek, 2021-08-25:
I believe that you would have to change the strategy to achieve your goals. Instead of handling two different motors with a single driver you should use two separate drivers, one for each motor. You can run two instances of Astroberry Focuser and assign unique names to each of them. This way you can have two focusers in your system, each with its own configuration.
Implementing the feature you requested is a "no-go"

jctk, 2021-08-25:
Thanks for the advice.
I understand that it will not be implemented.

I'm going to read the indi documentation you gave me.
Maybe I need to learn how to do this with Indi Web Manager or Kstars.