rkaczorek/astroberry-diy #24: Help : usage of astroberry-diy with Ubuntu 20.04

pcevey, 2021-12-09:
Hi,
I need help for astroberry-diy please.
I am using KStars/ekos on my Raspberry PI4, running with Ubuntu 20.04 without problem (connected to telescope, mount, camera, guiding camera).
I tried to install astroberry-diy (V2.9), mainly for astroberry-focuser, using your instructions. After connecting all pins to a step motor driver (Big Easy Driver from Sparkfun, with A4988 IC), it is visible on Indi control panel, with 3 tabs (Otions, Main control, General Infos), but I get an error at connexion : Problem initiating Astroberry Focuser. 
I am not a computer specialist, therefore I send you print screens of the compilation and installation process for diagnostic : can you tell me if I can compile and use astroberry-diy with Ubuntu 20.04 and if there are special configuration to accomplish to have it work?
Thanks a lot!
[astroberry_diy_compil_term_printscreen.odt](https://github.com/rkaczorek/astroberry-diy/files/7684275/astroberry_diy_compil_term_printscreen.odt)

pcevey, 2021-12-15:
Thanks.
I did gpiod installation (see first

pcevey, 2021-12-15:
Sorry, bad manipulation.
I did gpiod installation (see first printscreen of synaptic with all gpio packages installed or not).
The result is the same: astroberry focuser is not found by ekos (see 2nd print screen). And I checked again the GPIO connections, the motor connection and the voltage source (9V battery).

May  I kindly ask you to help me again and tell me how I can diagnose where the problem is please ?
Thanks in advance,
Kind r
[image]
[image]
Kind regards,
Pierre

pcevey, 2021-12-15:
To be complete, I have to add that I have a RTC connected to GPIO2 and 3 + a fan on 5V and ground (pins 2 or 4 and 6).

rkaczorek, 2021-12-15:
Please provide output of these two commands:
1) `gpiodetect`
2) `ls -al /dev/gpiochip0`

pcevey, 2021-12-15:
ubuntu@rpi4-ubuntu20-04:~$ sudo gpiodetect
gpiochip0 [pinctrl-bcm2835] (54 lines)
gpiochip1 [raspberrypi-exp-gpio] (8 lines)
ubuntu@rpi4-ubuntu20-04:~$ ls -al /dev/gpiochip0
crw------- 1 root root 254, 0 déc 15 20:05 /dev/gpiochip0

rkaczorek, 2021-12-16:
Now it's clear. The problem is that your GPIO device /dev/gpiochip0 is owned and accessible only by root. It should be accessible by user you start astroberry-diy driver, but it is not on you system. Raspberry OS / Raspbian / Astroberry sets permissions to these device this way `crw-rw---- 1 root gpio 254, 0 Nov  5 12:33 /dev/gpiochip0` so the device is accessible (rw) by group gpio. In Astroberry Server the astroberry user is member of gpio group so everything works ok.
Quick solution is to change permissions to the device to rw for everyone i.e. `sudo chmod 666 /dev/gpiochip0` however it will not be persistent between reboots. To make it persistant you need to add an udev rule for /dev/gpiochip0. You can easily find examples on the Internet.

pcevey, 2021-12-16:
ok ! this works ! Thanks a lot !
Concerning the udev rule, I am a bit uncomfortable to do it without aksing you if adding a rule (in a numbered file) to /etc/udev/rules.d which simply says : NAME=="/dev/gpiochip0", GROUP:="dialout", MODE="666"
N.B.: The group name comes from other recommendations I read when trying to understand the functioning of the gpio pins and I am part of this group.
Also the /etc/udev/rules.d directory contains already a rule file which I sent in attachment. Could it cause a problem with the rule I add and is the order between this file and mine important ?

Again. thanks a lot for your help!
[image]

pcevey, 2021-12-17:
Found my way ! udev rule established and stored, tested after reboot ok!
Thanks for your help!

closed 2021-12-17 by pcevey