rkaczorek/astroberry-diy #26: Standby timer not reset by new command

keith-b-git, 2022-01-28:
If standby is enabled and a a move command is given and then a new move command is given within the timer window, standby from the first command will still be activated, possibly preventing the correct execution of later commands.

To reproduce- 
Enable standby and set time to 60s.
Execute a step command.
Execute a second command after 50s

expected result - the motor should not go into standby until 60s after the second command
actual result - the motor goes into standby 60s after the first command and will try to go into standby again 60s after the 2nd, and this  can disrupt a move in process at these times.

Thanks for producing this useful piece of software,

Keith

rkaczorek, 2022-01-29:
Thanks for reporting. I will take look at it

rkaczorek, 2022-01-29:
The fix should be pushed to repo in an hour or so. Test and report back.

closed 2022-01-29 by rkaczorek

keith-b-git, 2022-01-29:
Thanks for your quick response, tested and OK now.

commit bce421f1 (2022-01-29): Fixes #26