rkaczorek/astroberry-diy #27: Joystick axis support

teoteo, 2022-02-07:
If I connect a Joystick or gamepad, I can see [3 buttons to focus in](https://www.indilib.org/forum/focusers-filter-wheels/11240-astroberry-diy-focuser-and-joystick.html#80274), focus out and abort focus
[image]

What about adding the ability to use the analog axis of joystick to control the focus speed? 
[image]

Axis close to 0 = abort focusing
Axis between minimum and maximum , a ramp related to relative minimum and maximum speed set.

rkaczorek, 2022-02-07:
This should be reported here https://github.com/indilib/indi

teoteo, 2022-02-07:
OK, thanks for the suggestion