rkaczorek/astroberry-diy #33: CH42 board

NorthBaandee, 2022-11-11:
Is it possible to use this board which has only STEP DIR and EN inputs ?
https://www.ebay.com/itm/42-CH-Stepper-Motor-Driver-Expansion-Board-DRV8825-A4988-for-3D-Printer-UNO-R3-/132540586828

rkaczorek, 2022-11-13:
You could use any board with STEP, DIR and EN inputs only. 
However this one is not suitable for Raspberry Pi as it uses logic voltage 5V (RPi requires 3.3V) and it it requires stepper motor microcontroller - it is just an expansion board, which will do nothing on its own. You can overcome the first issue using logic voltage shifters 5V->3.3V but it is probably unnecessary complication. I would look for something else.

NorthBaandee, 2022-11-13:
I have put this board in the socket 
https://www.ebay.com.au/itm/155104238416 but still no good??